pyavitz/debian-image-builder #1: bcm2711-*.img missing

fdcon, 2021-03-01:
Have tried to build feature branch but get this error all the time

pyavitz, 2021-03-01:
I am going to assume that you are using the `dialog` menu interface?
Did you install all of the needed dependencies?
What OS are you running?
Are you using docker?

I am running a build now to test if this error presents its self.

fdcon, 2021-03-01:
Hi,
Yes using the dialog menu interface. Have tried several different ways. Both using docker build and cross compile on both Debian (buster) and ubuntu 20.04.

Thanks

pyavitz, 2021-03-01:
You must be missing dependencies?

[image]

Whilst inside debian-image-builder run:
`make ccompile` cross compile 
`make ncompile` native compile 

You can test this by editing the userdata.txt file.
Change: verbose=0 to verbose=1 (this will show errors in the stage1 script)
Run each command individually: `make raspi4-uboot` `make raspi4-kernel` `make rootfs` `make raspi4-image`

fdcon, 2021-03-02:
Hi,
It wasn’t the dependencies. It was that I needed to do 
Make update

Thats the step that I missed.
Thanks for looking at it.

Sorry for missing that step.

f.

> On 2 Mar 2021, at 00:16, Patrick <notifications@github.com> wrote:
> 
> make ccompile cross compile
> make ncompile native compile

closed 2021-03-02 by fdcon

pyavitz, 2021-03-02:
Ah yes, the docker builder. Happy you figured it out.