pyavitz/debian-image-builder #4: M1 Mini

wcatz, 2021-06-30:
Hi guys

This is a cool project. I was trying to get 5.13 kernel to build, in an attempt to boot Apples M1 mini when I came across this.

5.13 not supported yet with hirsute? Thoughts on the M1.

Going to give it another go. 💯

ptr1337, 2021-06-30:
I got a M1-Macbook, so i dont know how far the GPU Drivers are, but i would give also a try.

How are you trying to create the install right now? With Qemu ?

pyavitz, 2021-06-30:
The builder currently builds what I either own or what people have given me information on. As for 5.13.y and above, I pretty much stopped at 5.10 LTS as this is pretty much a one man show. Finding, creating and porting patches for the kernels, is in its self a job :)

I honestly know nothing about the M1 mini, but I'm going to assume I would need to create a packaging patch for it, so the kernel would be of use. As for the boot process, I know nothing about it.

Beyond that the builder would need to know about the SoC it is building for, so once the kernel has been created the img could be thrown together.

wcatz, 2021-06-30:
Using Correlium's hacked together image in production for a few months but now that support has started (5.13)was trying to see if I could get latest Ubuntu server to boot and at first glance this project has some potential. 

This chip is incredibly fast and the nvme drive has a read speed of 25gps write of 1.3gps

[https://corellium.com/blog/linux-m1](url)
[https://www.omgubuntu.co.uk/2021/06/new-linux-5-13-kernel-features](url)

I am interested in it as Ubuntu server gpu, wireless doesnt matter to me. Though wireless works fine.

Thanks for getting back so quick. My knowledge on the matter is limited but willing to test any ideas you may have.
best regards

ptr1337, 2021-07-08:
Lima could be solution for you.

closed 2021-08-14 by pyavitz