pyavitz/debian-image-builder #25: debian-image-builder vs rpi-img-builder

djsftree, 2022-05-17:
Which do you suggest for Debian / RPI4?

pyavitz, 2022-05-17:
The pi builder pulls its kernel from the [Raspberry Pi Github](https://github.com/raspberrypi/linux), so it benefits from the work they put into it. Bug fixes, defconfigs. code changes and what have you? I would say the pi builder is better for those who want that support.

The Pi4 was added to this builder to watch the progression of mainline Linux on the board more than anything else. Although last I checked it works fine `on a headless level`, I wouldn't use it for production. Unless you are trying to do something specific that is and don't require the Foundations efforts in that Linux space.

djsftree, 2022-05-17:
Understood, many thanks!

closed 2022-05-17 by djsftree