pyavitz/debian-image-builder #38: Request for standard Github payment processor for sponsorship (ie, not Paypal)

em-winterschon, 2022-09-18:
I'd like to add this project to my list of monthly github developer donation/sponsorships but using paypal is not a relevant option. Offering the built-in github sponsorship payment processor would be preferred. 

Thanks!

pyavitz, 2022-09-18:
Thank you. I'll look into it and let you know if and when its set up.

pyavitz, 2023-05-17:
Unless something changes in the future, I won't be opening an exclusive account to receive donation/sponsorships. Accounts aren't free and although a sponsorship would be great, there is no guarantee that it won't eventually dry up and leave me with the expense of having the account. Currently Paypal is the best option for me. I'm sorry it isn't for you, but I do very much appreciate your interest in the project.

closed 2023-05-17 by pyavitz