pyavitz/debian-image-builder #39: Nanopi R5S - keyboard configuration

Averell7, 2022-09-27:
defconfig ->  rockchip64-rk356x_defconfig 

using 
rk3568-nanopi-r5s-debian-bullseye-5.19.11-arm64-2022-09-26.img.xz 
I am unable to set the keyboard configuration. The config menu works fine, but at load time, there is an error :
FAILED : Failed to start Set console font and keymap. 
And the keybaord layout is not changed.

pyavitz, 2022-09-27:
try running; `sudo systemctl restart console-setup`

pyavitz, 2022-09-27:
Good. I've never personally had this issue before, but if it should happen again, I'll add it. Thanx.

Averell7, 2022-09-28:
The error continues on startup, and there is surely something to tune : 

Sep 28 02:59:51 nanopir5s console-setup.sh[348]: /usr/bin/setupcon: 870: cannot open /tmp/tmpkbd.rBhw5A: No such file
Sep 28 02:59:51 nanopir5s systemd[1]: console-setup.service: Main process exited, code=exited, status=1/FAILURE
Sep 28 02:59:51 nanopir5s mariadbd[772]: 2022-09-28  2:59:51 0 [Note] /usr/sbin/mariadbd (mysqld 10.5.15-MariaDB-0+deb11u1) starting as process 772 ...
Sep 28 02:59:51 nanopir5s systemd[1]: console-setup.service: Failed with result 'exit-code'.
Sep 28 02:59:51 nanopir5s systemd[1]: Failed to start Set console font and keymap.

Averell7, 2022-09-28:
I tried to run /usr/bin/setupcon on the command line : it does not return an error.

pyavitz, 2022-09-28:
Read through this and see if it helps; https://github.com/MichaIng/DietPi/issues/2912

pyavitz, 2022-09-28:
If all else fails, see if you get the desired results with Ubuntu Jammy; https://github.com/pyavitz/binary/releases/tag/images

Averell7, 2022-09-28:
OK, I'll test. Leave me some time since this is not critical.

Averell7, 2022-10-07:
I used another time the solution you proposed above : 
`systemctl restart console-setup.service`
I don't know why this time it worked. It is now several days (and a lot of restarts) I have done it and no problem. 
Maybe some commands I used from the link you gave fixed something ?

closed 2022-10-07 by Averell7