pyavitz/debian-image-builder #40: Nanopi R5S - ethernet ports order

Averell7, 2022-09-27:
In the images offered by the constructor, the order of eth0, eth1, eth2 is :

eth0 = WAN  (right)
eth1 = LAN1  (middle)
eth2 = LAN2  (left)

using rk3568-nanopi-r5s-debian-bullseye-5.19.11-arm64-2022-09-26.img.xz
we have :

eth0 = LAN1 (middle)
eth1 = LAN2 (left)
eth2 = WAN (right)

this creates some confusion, because the order of ports from right to left (R5S box is designed like that) is eth2, eth0, eth1. 

The choice of the constructor was better.

pyavitz, 2022-09-28:
Apply and reboot and let me know what happens
```sh
sudo su
echo 'SUBSYSTEM=="net", ACTION=="add", DRIVERS=="?*", ATTR{address}=="00:00:00:00:00:00",ATTR{dev_id}=="0x0", ATTR{type}=="1", KERNEL=="eth2", NAME="eth0"' > /etc/udev/rules.d/10-nanopi-r5s-eth.rules
echo 'SUBSYSTEM=="net", ACTION=="add", DRIVERS=="?*", ATTR{address}=="00:00:00:00:00:00",ATTR{dev_id}=="0x0", ATTR{type}=="1", KERNEL=="eth0", NAME="eth1"' >> /etc/udev/rules.d/10-nanopi-r5s-eth.rules
echo 'SUBSYSTEM=="net", ACTION=="add", DRIVERS=="?*", ATTR{address}=="00:00:00:00:00:00",ATTR{dev_id}=="0x0", ATTR{type}=="1", KERNEL=="eth1", NAME="eth2"' >> /etc/udev/rules.d/10-nanopi-r5s-eth.rules
```

Averell7, 2022-09-28:
Thank you for the proposal. I tested but there is no change. 
Since I have two R5S running, one with the FriendlyElec system (Ubuntu 22) and the other with your build, I tried to understand how they obtain the interface order they have but I found nothing. I don't see anything relevant in /usr/lib/udev/rules.d and in /etc/udev/rules.d I find a strange 70-persistent-net.rules which is empty. I tried to create in the Armbian system, but it did not change anything (no surprise, since there is no corresponding rule in /usr
If you have any test which can be useful on the two devices, tell me.

pyavitz, 2022-09-28:
I don't actually have the HW, so I can't really test anything on this front. I did check the udev rule against another board and the eth port number changed, but thats as far as I can go with it. I'm sure a solution will eventually present its self.

pyavitz, 2022-09-28:
9bx154, suggested something like this:
```sh
echo 'SUBSYSTEM=="net", ACTION=="add", DRIVERS=="rk_gmac-dwmac", KERNEL=="eth2", NAME="eth0"' > /etc/udev/rules.d/10-nanopi-r5s-eth.rules
echo 'SUBSYSTEM=="net", ACTION=="add", DRIVERS=="r8169", KERNEL=="eth0", NAME="eth1"' >> /etc/udev/rules.d/10-nanopi-r5s-eth.rules
echo 'SUBSYSTEM=="net", ACTION=="add", DRIVERS=="r8169", KERNEL=="eth1", NAME="eth2"' >> /etc/udev/rules.d/10-nanopi-r5s-eth.rules

```

Averell7, 2022-09-28:
The suggestion of 9bx154 is good. It works fine if I rename to different names. But unfortunately it creates confusion when the same names are used. Since I don't need eth2 presently, I did this : 

SUBSYSTEM=="net", ACTION=="add", DRIVERS=="r8169", KERNEL=="eth1", NAME="xth2"
SUBSYSTEM=="net", ACTION=="add", DRIVERS=="r8169", KERNEL=="eth0", NAME="eth1"
SUBSYSTEM=="net", ACTION=="add", DRIVERS=="rk_gmac-dwmac", KERNEL=="eth2", NAME="eth0"

It remains a dirty workaround, but for now it is working.

Averell7, 2022-09-28:
Probably the good solution will be to tweak systemd so that rk_gmac-dwmac is handled before r8169. 
I will look in that direction and get you informed
Thanks.

pyavitz, 2022-09-28:
or maybe turn eth2 into wan0?

Averell7, 2022-10-07:
The solution was in two steps : 
```
SUBSYSTEM=="net", ACTION=="add", DRIVERS=="rk_gmac-dwmac", NAME="WAN"
SUBSYSTEM=="net", ACTION=="add", DRIVERS=="r8169", KERNEL=="eth0", NAME="LAN1"
SUBSYSTEM=="net", ACTION=="add", DRIVERS=="r8169", KERNEL=="eth1", NAME="LAN2"
```
This could be a good default for this device, to have the port names similar to what is printed on the box. 

And after that, I use /etc/network/interfaces to rename the interfaces to what I need. 
```
rename WAN=eth0
rename LAN1=eth1
```

closed 2022-10-07 by Averell7