pyavitz/debian-image-builder #45: Can't get the final build

FillHK, 2022-11-09:
All the steps are done, with no issues but after the command:
`make menu`

I see: 
```
ls: cannot access 'meson-xxx*.img': no such file or direct 
debian-bullseue-riitfs-arm64.tar.xz

output/odroidn2/:
```
I have this directory but it is empty

pyavitz, 2022-11-09:
What distro and release are you using as a Host?
Did you install all the needed dependencies?

---

If there is nothing in the output directory, there is a problem.

You should have something like this:
```sh
odroid: debian-image-builder git:( feature ?:23 ✗ ) $ ls output/odroidn2
linux-headers-meson-g12b_6.0.7-1_arm64.deb  u-boot.bin.sd.bin
linux-image-meson-g12b_6.0.7-1_arm64.deb    u-boot-meson-g12b_2022.10-1_arm64.deb
u-boot.bin
```
and a rootfs tarballs: `debian-bullseye-rootfs-arm64.tar.xz`

Try running the commands manually to see if an error presents its self.
```sh
make uboot board=odroidn2
make kernel board=odroidn2
make rootfs board=odroidn2
make image board=odroidn2
```
You can also turn on verbosity in the userdata.txt file: `verbose=1` 
If using the menu interface you can also enable logging.

FillHK, 2022-11-09:
So far next result with **make uboot board=odroidn2**: 

```
SHIPPED dts/dt.dtb
  CAT     u-boot-dtb.bin
  COPY    u-boot.dtb
  COPY    u-boot.bin
  LD      u-boot.elf
Odroid N2: Firmware Image Package
2096+0 records in
2096+0 records out
2096 bytes (2.1 kB, 2.0 KiB) copied, 0.0128976 s, 163 kB/s
4701+0 records in
4701+0 records out
4701 bytes (4.7 kB, 4.6 KiB) copied, 0.0266801 s, 176 kB/s
9408+0 records in
9408+0 records out
9408 bytes (9.4 kB, 9.2 KiB) copied, 0.0547915 s, 172 kB/s
2384+0 records in
2384+0 records out
2384 bytes (2.4 kB, 2.3 KiB) copied, 0.0154244 s, 155 kB/s
lib/function/fip: line 260: /userdata/debian-image-builder-feature/sources/u-boot/fip/aml_encrypt_g12b: cannot execute binary file: Exec format error
lib/function/fip: line 263: /userdata/debian-image-builder-feature/sources/u-boot/fip/aml_encrypt_g12b: cannot execute binary file: Exec format error
lib/function/fip: line 266: /userdata/debian-image-builder-feature/sources/u-boot/fip/aml_encrypt_g12b: cannot execute binary file: Exec format error
lib/function/fip: line 269: /userdata/debian-image-builder-feature/sources/u-boot/fip/aml_encrypt_g12b: cannot execute binary file: Exec format error
lib/function/fip: line 272: /userdata/debian-image-builder-feature/sources/u-boot/fip/aml_encrypt_g12b: cannot execute binary file: Exec format error
lib/function/fip: line 274: /userdata/debian-image-builder-feature/sources/u-boot/fip/aml_encrypt_g12b: cannot execute binary file: Exec format error
Something went wrong? Please report.
```

FillHK, 2022-11-09:
Started from first steps:
1. **make ncompile**
```
root@HomeAssistant:/userdata/debian-image-builder-feature# make ncompile
# Installing native dependencies:

make: *** [Makefile:110: ncompile] Error 100
```

FillHK, 2022-11-09:
I see why it happens (i assume)
`(28: No space left on device)`

I have one device with Debian installed only, and my targeted platform is Odroid N2.
Seems my OdroidN2 has no Debian installed I can't execute the code directly on a device...
Have no idea now how I can build an image for Odroid arm64

pyavitz, 2022-11-09:
I can build you an img if you like? This is for the `original` Odroid N2?

pyavitz, 2022-11-09:
It looks like you are either missing dependencies or not enough space on the unit.

Here is an [Odroid N2](https://github.com/pyavitz/binary/releases/download/images/meson-g12b-odroid-n2-debian-bullseye-6.0.7-arm64-ext4-2022-11-09-1331.img.xz) image. You need to follow these [instructions](https://github.com/pyavitz/debian-image-builder#bootrename_to_useraccounttxt) for creating a user account upon first boot. If you choose not to use the `useraccount.txt` file, make sure the unit has a monitor and keyboard connected to it.

FillHK, 2022-11-09:
I used balena ether for your image.
After start - nothing happens - blank screen ;(

pyavitz, 2022-11-09:
> I think I have Odroin N2+

Here is a N2+ image: https://github.com/pyavitz/binary/releases/download/images/meson64_odroidn2_plus-debian-bullseye-6.0.7-arm64-ext4-2022-11-09-1444.img.xz

Both images are meant to be flashed to SDCARD. After booting from SD, if you have an eMMC you wanna transfer the install too; attach the module and run: `sudo write2mmc`

FillHK, 2022-11-10:
Done with Debian, works.

Just one issue left with HomeAssistant:
`Unsupported system - CGroup version`

Have re-run the installation script many times but with no luck.

pyavitz, 2022-11-10:
Ok. So reviewing the script its self, this should be appended to ur extlinux.conf file during the install.
https://github.com/pyavitz/debian-image-builder/blob/feature/files/scripts/menu-config#L304

If you need to add something to the boot command line before hand, you can do so by manually adding it to `/boot/extlinux/extlinux.conf`

FillHK, 2022-11-10:
You made my day today!!
```
Host Operating System	Debian GNU/Linux 11 (bullseye)
Update Channel	stable
Supervisor Version	supervisor-2022.11.dev0802
Agent Version	1.4.1
Docker Version	20.10.21
Disk Total	114.6 GB
Disk Used	5.6 GB
Healthy	true
Supported	true
Supervisor API	ok
Version API	ok
```

Appreciate your help and support!

pyavitz, 2022-11-10:
Did that not get appended during the install?

pyavitz, 2022-11-10:
Beyond the usual limitations of the N2 and USB, I know of no issues. I'll ask CPmentor if he knows anything. He wrote the install part of the script and uses HA.

FillHK, 2022-11-10:
I used this device on this hardware before, sure it's supported...

pyavitz, 2022-11-10:
[Zigbee2MQTT](https://www.zigbee2mqtt.io/)

Does this require thirdparty kernel support?

0n3man, 2022-11-10:
I use [ZHA](https://www.home-assistant.io/integrations/zha) on my HA system for zigbee communications.  At one location I use a GoControl HUSBZB-1 and another I use a CronBee II.  Both work fine on the odroid N2+.  Can you give ZHA a try and see if detects your USB device?

0n3man, 2022-11-11:
[FillHK](https://github.com/FillHK) if you got to debian-image-build from the home assistant forum, possible [here](https://community.home-assistant.io/t/how-to-supervised-ha-install-on-odroid-n2-and-others-with-clean-debian-11/428675).  Can you add something to that HA thread to pop it to the top so others will see you had success going this route.  Thanks

jasonmadigan, 2022-11-24:
> Here is a N2+ image: https://github.com/pyavitz/binary/releases/download/images/meson64_odroidn2_plus-debian-bullseye-6.0.7-arm64-ext4-2022-11-09-1606.img.xz
>
> Both images are meant to be flashed to [...]

These images seem to be 404ing.

pyavitz, 2022-11-25:
yeah sorry I generally don't release imgs -> https://github.com/pyavitz/binary/releases/tag/images