pyavitz/debian-image-builder #47: how to set kernel-parameters

Ventusfahrer, 2022-11-16:
I used this projekt to build a debian bullseye image for Home Assistant for an Odroid N2+ board (mmc installation).

Everything works fine except that CGroups V2 is default in Bullseye and HomeAssistant did not support V2 (because of Docker??). Getting the necessary kernel-parameter set in the boot-options (systemd.unified_cgroup_hierarchy=false needs to be set) causes me headaches at the moment (I did not get managed it at all up to now).

It would be nice to have it enabled by an option in your configuration would be very helpfull!

kind regards

Peter

0n3man, 2022-11-16:
Did you do the HA install using the menu-config, as it includes ability to do full HA setup on top of the image?

pyavitz, 2022-11-16:
If for whatever reason you didn't use it, those changes can be made by editing; `sudo nano /boot/extlinux/extlinux.conf`

Ventusfahrer, 2022-11-17:
> Did you do the HA install using the menu-config, as it includes ability to do full HA setup on top of the image?

yes, I used the menu-config Software/HA-Installation step to install an setup home assistant. 

There was another issue with the operating system and I am currently not shure where to place the issue:
iptables and ip6tables do need an
'''
update-alternatives --set iptables /usr/sbin/iptables-legacy
update-alternatives --set ip6tables /usr/sbin/ip6tables-legacy
'''

in order to work. Without it, docker-ce startup will fail

Ventusfahrer, 2022-11-17:
> If for whatever reason you didn't use it, those changes can be made by editing; `sudo nano /boot/extlinux/extlinux.conf`

really?

the directory /boot/extlinux does not exist on the file system.

therefore I don't know about the file structure of extlinux.conf

What it found was /boot/boot.cmd. I tried to use it and put systemd.unified_cgroup_hierarchy=false into the line

setenv extra "systemd.unified_cgroup_hierarchy=false"

But just rebooting won't do the job. Googling, I found some hints that you have to run mkimage in order to get an updated /boot/boot.scr. But after running mkimage, boot.scr was updated and nothing changed on the /proc/cmdline. 

The last try was to run flash_uboot.sh. After that the operating system refused to load and I reinstalled the whole stuff.

Last shot was to change /boot/config.ini . This one looks more reasonable for the current kernel parameters shown in /proc/cmdline. But again it is not sufficient to only change the file. I need also to find out the right tool-chain to get the changes to the active boot-command.

Any help ist highly appreciated!

Thanks 

Peter

Ventusfahrer, 2022-11-17:
curious - that's exactly what I found out in parallel. Fine, so far.

The system was setup by installing the image following: https://magazine.odroid.com/article/os-installation-using-petitboot-and-usb-otg/

I am using the **Direct Boot** option.

My problem was, that for the first try I was using the manual description of https://github.com/home-assistant/supervised-installer. In this case /boot/config.ini is not created at all.

Running the step:

```
wget https://github.com/home-assistant/supervised-installer/releases/latest/download/homeassistant-supervised.deb
dpkg -i homeassistant-supervised.deb
```
will end up in the log:

```
root@7vhas:~# dpkg -i homeassistant-supervised.deb
(Lese Datenbank ... 56046 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von homeassistant-supervised.deb ...
.
.
.
--- checkonline.home-assistant.io ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 19.733/19.733/19.733/0.000 ms
[info] Install supervisor startup scripts
[info] Install AppArmor scripts
[info] Start Home Assistant Supervised
[info] Installing the 'ha' cli
[warn] Could not find /etc/default/grub or /boot/firmware/cmdline.txt failed to switch to cgroup v1
[info] Within a few minutes you will be able to reach Home Assistant at:
[info] http://homeassistant.local:8123 or using the IP address of your
[info] machine: http://192.168.221.201:8123
root@7vhas:~#
```
The warning pointing to the grub configuration points out, that the package is not able to fix cgroup v1.

When I run the HA installation via menu-config cgroup v1 is still not configured, but /boot/config.ini is there. 

Configuration after menu-config installed HA:

peter@7vhas:~$ more /boot/config.ini
```
[user_options]
bootlabel=Amlogic
uuid=PARTUUID=06233f8d-02
rootfstype=rootfstype=ext4
verbose=1
usbquirks=usb-storage.quirks=0x1f75:0x0621:u
extra="no_console_suspend clk_ignore_unused"
console="console=tty1 console=ttyAML0,115200n8 console=both"

fk_kvers=Image
initrd=initrd.gz
kernel_addr_r=0x10800000
ramdisk_addr_r=0x30800000
fdtdir=amlogic
fdtfile=meson64_odroidn2_plus.dtb
fdt_addr_r=0x20000000

; Device Tree Overlay
overlay_resize=16384
overlay_profile=
; overlays="spi0 i2c0 i2c1 uart0"

```

```
peter@7vhas:~$ stat -fc %T /sys/fs/cgroup/
cgroup2fs

```
rerun HA deb package to fix cgroup v1

partial output: 
```
[info] Start Home Assistant Supervised
[info] Installing the 'ha' cli
[info] Switching to cgroup v1
got it
[info] Within a few minutes you will be able to reach Home Assistant at:
[info] http://homeassistant.local:8123 or using the IP address of your
[info] machine: http://192.168.221.201:8123
[warn] A reboot is required to apply changes to grub.
```
suggests, that something has changed but /boot/config.ini is not changed and cgroups still v2 configured.

I still have manually to add 
systemd.unified_cgroup_hierarchy=false 
to /boot/config.ini
either to 
`extra="no_console_suspend clk_ignore_unused systemd.unified_cgroup_hierarchy=false"`
or 
`console="console=tty1 console=ttyAML0,115200n8 console=both systemd.unified_cgroup_hierarchy=false"`
line.

So homeassistant-supervised.deb should also see an improvement?

Ventusfahrer, 2022-11-17:
...maybe it is worth to mention details of the step in menu-config - Software- HA Installation which creates /boot/config.ini somewhere in the documentation so one is able to run it manually?

0n3man, 2022-11-17:
I just did verification that installing with uboot on emmc, everything works with these two lines in place using the latest LTS kernel
```
update-alternatives --set iptables /usr/sbin/iptables-legacy
update-alternatives --set ip6tables /usr/sbin/ip6tables-legacy
```
Something most of changed in the LTS kernel as these were not previously required.

I need to do another install to fully verify that as an alternative one can use the 6.0.9 kernel.  I ran that test last night, but I had set the first of the two lines recommended above.  

I could look at modifying the HA install script to update the /boot/config.ini file in cases where petitboot has been selected as the boot loader and also add in these two lines if the LTS kernel is being used.

Ventusfahrer, 2022-11-21:
> The HA install script now checks for petitboot and presents a notice that the following kernel params should be added to the config.ini.
> 
> ```shell
> systemd.unified_cgroup_hierarchy=false systemd.legacy_systemd_cgroup_controller=false usbcore.autosuspend=-1 usbcore.autosuspend=-1 clk_ignore_unused cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory apparmor=1 security=apparmor
> ```
> 
> If it becomes a more popular option "petitboot" in the future, I will automate the process.

I tested your suggested parameter line, but did not succeed:

here is my config.ini:

```
[user_options]
bootlabel=Amlogic
uuid=PARTUUID=06233f8d-02
rootfstype=rootfstype=ext4
verbose=1
usbquirks=usb-storage.quirks=0x1f75:0x0621:u
;extra="no_console_suspend clk_ignore_unused systemd.unified_cgroup_hierarchy=false"
;extra="no_console_suspend clk_ignore_unused systemd.unified_cgroup_hierarchy=false  systemd.legacy_systemd_cgroup_controller=false"
extra="systemd.unified_cgroup_hierarchy=false systemd.legacy_systemd_cgroup_controller=false usbcore.autosuspend=-1 clk_ignore_unused cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory apparmor=1 security=apparmor"
console="console=tty1 console=ttyAML0,115200n8 console=both"

fk_kvers=Image
initrd=initrd.gz
kernel_addr_r=0x10800000
ramdisk_addr_r=0x30800000
fdtdir=amlogic
fdtfile=meson64_odroidn2_plus.dtb
fdt_addr_r=0x20000000

; Device Tree Overlay
overlay_resize=16384
overlay_profile=
; overlays="spi0 i2c0 i2c1 uart0"

```
mount -l | grep cgroup 

shows:

`cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime)
`

When 

```
extra="no_console_suspend clk_ignore_unused systemd.unified_cgroup_hierarchy=false systemd.legacy_systemd_cgroup_controller=false"
```
is active, 

`mount - l | grep cgroup 
`

shows the right configuration:

````
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,size=4096k,nr_inodes=1024,mode=755)
cgroup2 on /sys/fs/cgroup/unified type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,name=systemd)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/rdma type cgroup (rw,nosuid,nodev,noexec,relatime,rdma)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,nosuid,nodev,noexec,relatime,pids)

```

pyavitz, 2022-11-21:
Update the kernel and see if it makes any difference.

```sh
wget https://github.com/pyavitz/binary/releases/download/060420/linux-headers-meson-g12b_5.15.79-1_arm64.deb
wget https://github.com/pyavitz/binary/releases/download/060420/linux-image-meson-g12b_5.15.79-1_arm64.deb
sudo dpkg -i linux-*.deb
```

Ventusfahrer, 2022-11-21:
Installation  complanes about downgrading the kernel version :-)
But the version number is one I like more thant th 6.0.8, which looks very strange, if you are used to the numbering schema on ArchLinux or Ubuntu...

```
dpkg: warning: downgrading linux-headers-meson-g12b from 6.0.8-1 to 5.15.79-1
(Reading database ... 56106 files and directories currently installed.)
Preparing to unpack linux-headers-meson-g12b_5.15.79-1_arm64.deb ...
Unpacking linux-headers-meson-g12b (5.15.79-1) over (6.0.8-1) ...
dpkg: warning: downgrading linux-image-meson-g12b from 6.0.8-1 to 5.15.79-1
Preparing to unpack linux-image-meson-g12b_5.15.79-1_arm64.deb ...
Unpacking linux-image-meson-g12b (5.15.79-1) over (6.0.8-1) ...
Setting up linux-headers-meson-g12b (5.15.79-1) ...
Compiling headers ...
^Cdpkg: error processing package linux-headers-meson-g12b (--install):
 installed linux-headers-meson-g12b package post-installation script subprocess was interrupted
Setting up linux-image-meson-g12b (5.15.79-1) ...
update-initramfs: Generating /boot/initrd.img-5.15.79
Generating uInitrd for U-Boot:
Image Name:   initramfs-5.15.79
Created:      Mon Nov 21 15:28:18 2022
Image Type:   ARM Linux RAMDisk Image (gzip compressed)
Data Size:    6966581 Bytes = 6803.30 KiB = 6.64 MiB
Load Address: 00000000
Entry Point:  00000000
Creating initrd.gz.
Errors were encountered while processing:
 linux-headers-meson-g12b

```
the second try installing the packages did succeed.

But the behaviour does not change with the provided kernel:

```
peter@7vhas:~$ cat /proc/cmdline 
console=tty1 console=ttyAML0,115200n8 console=both ro root=PARTUUID=06233f8d-02 net.ifnames=0 rootfstype=ext4 fsck.repair=yes loglevel=1 "systemd.unified_cgroup_hierarchy=false systemd.legacy_systemd_cgroup_controller=false usbcore.autosuspend=-1 clk_ignore_unused cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory apparmorextra usb-storage.quirks=0x1f75:0x0621:u rootwait
```
```
peter@7vhas:~$ uname -a
Linux 7vhas 5.15.79 #1 SMP PREEMPT Mon Nov 21 06:44:34 EST 2022 aarch64 GNU/Linux

```
```
peter@7vhas:~$ cat /proc/cmdline
console=tty1 console=ttyAML0,115200n8 console=both ro root=PARTUUID=06233f8d-02 net.ifnames=0 rootfstype=ext4 fsck.repair=yes loglevel=1 "systemd.unified_cgroup_hierarchy=false systemd.legacy_systemd_cgroup_controller=false usbcore.autosuspend=-1 clk_ignore_unused cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory apparmorextra usb-storage.quirks=0x1f75:0x0621:u rootwait
```

```
peter@7vhas:~$ mount -l | grep cgroup
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime)

```

pyavitz, 2022-11-21:
One kernel version is LTS and the other Stable is all that is.

This is odd:
```sh
^Cdpkg: error processing package linux-headers-meson-g12b (--install):
 installed linux-headers-meson-g12b package post-installation script subprocess was interrupted
```
But if it worked for you the second time, I guess nothing to worry about.

As for the cgroup issue you are still having, for me it resolved its self, after I updated the defconfig(s) and installed a new kernel.

cmdline typo:
wrong: `"systemd.unified_cgroup_hierarchy=false`
correct: `systemd.unified_cgroup_hierarchy=false`

pyavitz, 2022-11-23:
I just did a fresh install with petitboot and I'm not seeing anything stand out as en error.

```sh
patrick@odroidn2plus:~$ mount -l | grep cgroup
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,size=4096k,nr_inodes=1024,mode=755)
cgroup2 on /sys/fs/cgroup/unified type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,name=systemd)
cgroup on /sys/fs/cgroup/net_cls,net_prio type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls,net_prio)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,nosuid,nodev,noexec,relatime,pids)
cgroup on /sys/fs/cgroup/rdma type cgroup (rw,nosuid,nodev,noexec,relatime,rdma)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
```
```sh
patrick@odroidn2plus:~$ systemctl status
● odroidn2plus
    State: running
     Jobs: 0 queued
   Failed: 0 units
    Since: Tue 2022-11-22 18:57:17 EST; 3min 6s ago
   CGroup: /
           ├─docker 
           │ ├─0a3d09329322355c872bd2774bb57749a17dcdbf35f40f88740458283dd5092a 
           │ │ ├─1958 s6-svscan -t0 /var/run/s6/services
           │ │ ├─2044 s6-supervise s6-fdholderd
           │ │ ├─2214 s6-supervise coredns
           │ │ └─2218 coredns -conf /etc/corefile
           │ ├─9733eb3eb0da78a20a78c1275a0a444805a050dcfe2339988437c2e6bd1b220a 
           │ │ ├─4128 /package/admin/s6/command/s6-svscan -d4 -- /run/service
           │ │ ├─4150 s6-supervise s6-linux-init-shutdownd
           │ │ ├─4152 /package/admin/s6-linux-init/command/s6-linux-init-shutdown>
           │ │ ├─4159 s6-supervise s6rc-fdholder
           │ │ ├─4160 s6-supervise s6rc-oneshot-runner
           │ │ ├─4168 /package/admin/s6/command/s6-ipcserverd -1 -- /package/admi>
           │ │ ├─4196 s6-supervise home-assistant
           │ │ └─4198 python3 -m homeassistant --config /config
           │ ├─f9674c7888cbe8c4c88640bb4df746d10215a9a86150f7002c2ad0c8000701a0 
           │ │ ├─1337 /package/admin/s6/command/s6-svscan -d4 -- /run/service
           │ │ ├─1415 s6-supervise s6-linux-init-shutdownd
           │ │ ├─1418 /package/admin/s6-linux-init/command/s6-linux-init-shutdown>
           │ │ ├─1424 s6-supervise s6rc-fdholder
           │ │ ├─1425 s6-supervise s6rc-oneshot-runner
           │ │ ├─1433 /package/admin/s6/command/s6-ipcserverd -1 -- /package/admi>
           │ │ ├─1470 s6-supervise supervisor
           │ │ ├─1472 s6-supervise watchdog
           │ │ ├─1473 python3 -m supervisor
           │ │ ├─1474 bash /usr/bin/bashio ./run
           │ │ └─1496 sleep 300
           │ ├─ea9bb1419ef6dd426bf0c5ce0f81b38de7cd29b2eed2d61c16e22baa039e1093 
           │ │ ├─2887 s6-svscan -t0 /var/run/s6/services
           │ │ ├─2971 s6-supervise s6-fdholderd
           │ │ ├─3113 s6-supervise observer
           │ │ └─3117 observer
           │ ├─01942a7ba7e66832a69f71b43963b5cbf8b0c36e49ccd892466c508a4bc4bcc2 
           │ │ ├─3217 s6-svscan -t0 /var/run/s6/services
           │ │ ├─3258 s6-supervise s6-fdholderd
           │ │ ├─3407 s6-supervise mdns
           │ │ └─3411 mdns-repeater -f hassio
           │ ├─d93855e2de5d33cf96f1f2eb6220a8916737281f99aa68b64b4ab9f8926a1418 
           │ │ ├─2407 /package/admin/s6/command/s6-svscan -d4 -- /run/service
           │ │ ├─2472 s6-supervise s6-linux-init-shutdownd
           │ │ ├─2474 /package/admin/s6-linux-init/command/s6-linux-init-shutdown>
           │ │ ├─2479 s6-supervise s6rc-fdholder
           │ │ ├─2480 s6-supervise s6rc-oneshot-runner
           │ │ ├─2487 /package/admin/s6/command/s6-ipcserverd -1 -- /package/admi>
[... 346 lines cut ...]
bootlabel=Amlogic
uuid=PARTUUID=38320e9d-02
rootfstype=rootfstype=ext4
verbose=1
usbquirks=usb-storage.quirks=0x1f75:0x0621:u
extra="no_console_suspend clk_ignore_unused systemd.unified_cgroup_hierarchy=false systemd.legacy_systemd_cgroup_controller=false usbcore.autosuspend=-1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory apparmor=1 security=apparmor"
console="console=tty1 console=ttyAML0,115200n8 console=both"

fk_kvers=Image
initrd=initrd.gz
kernel_addr_r=0x10800000
ramdisk_addr_r=0x30800000
fdtdir=amlogic
fdtfile=meson64_odroidn2_plus.dtb
fdt_addr_r=0x20000000

; Device Tree Overlay
overlay_resize=16384
overlay_profile=
; overlays="spi0 i2c0 i2c1 uart0"
```
```sh
patrick@odroidn2plus:~$ dmesg | grep "Linux version"
[    0.000000] Linux version 5.15.79 (marvin@martian) (gcc-11 (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP PREEMPT Mon Nov 21 06:44:34 EST 2022
```

pyavitz, 2023-05-17:
This issue was opened back in `Nov 16, 2022` and it is now `May 16, 2023`. I last responded on `Nov 22, 2022` and have yet to get a reply.

closed 2023-05-17 by pyavitz