pyavitz/debian-image-builder #50: NanoPI R5S boot failure

electrified, 2022-11-30:
Hi there. Many thanks for your work on the builder.

The binary build in your other repository boots fine 
(rk3568-nanopi-r5s-debian-bullseye-5.19.12-arm64-2022-10-04.img), however images built by myself from the head of the feature branch (5772a1f5b6ee0b298b0c0fabedf7f1949ef3f555) falls over in UBoot.

[output.txt](https://github.com/pyavitz/debian-image-builder/files/10118316/output.txt)

[userdata.txt](https://github.com/pyavitz/debian-image-builder/files/10118329/userdata.txt)
[custom.txt](https://github.com/pyavitz/debian-image-builder/files/10118331/custom.txt)

I'm a beginner with U-Boot and Rockchip boards so trying to understand what the cause of the problem is.

pyavitz, 2022-11-30:
Sorry this is my bad, as its not properly documented. When building for RK3568 the current patch set in the builder only supports 5.19.y to 6.x.y. So latest STABLE releases.

Rebuild the kernel and use 6.0.10 instead of 5.15.80. In the future I'll see if I can add a switch that forces it to use the latest STABLE instead of the latest LTS.

pyavitz, 2022-11-30:
This commit will force pull the latest stable release. To override, the user will need to manually edit the lib/boards/$file and set the folowing variable to false: `FORCE_VERSION=""`

This isn't ideal, but considering only the current stable releases are supported it will have to do for now.

electrified, 2022-11-30:
Many thanks!

The defaults now produce a bootable image:

```
U-Boot SPL 2022.07 (Nov 30 2022 - 20:14:41 +0000)
Trying to boot from MMC2
INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-135-gcda1658bc-dirty:xsf
NOTICE:  BL31: Built : 14:19:26, Jun  1 2021
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    pmu v1 is valid
INFO:    dfs DDR fsp_param[0].freq_mhz= 1560MHz
INFO:    dfs DDR fsp_param[1].freq_mhz= 324MHz
INFO:    dfs DDR fsp_param[2].freq_mhz= 528MHz
INFO:    dfs DDR fsp_param[3].freq_mhz= 780MHz
INFO:    Using opteed sec cpu_context!
INFO:    boot cpu mask: 0
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0xa00000
INFO:    SPSR = 0x3c9

U-Boot 2022.07 (Nov 30 2022 - 20:14:41 +0000)

Model: FriendlyElec NanoPi R5S
DRAM:  rockchip_sdram_size fdc20208 10000201
rank 1 cs0_col 10 bk 3 cs0_row 17 bw 2 row_3_4 0
SDRAM base=0, size=100000000
3.7 GiB
Core:  232 devices, 17 uclasses, devicetree: separate
MMC:   mmc@fe2b0000: 1, mmc@fe310000: 0
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Model: FriendlyElec NanoPi R5S
Net:   No ethernet found.
Hit any key to stop autoboot:  0 
Card did not respond to voltage select! : -110
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:1...
Found /boot/extlinux/extlinux.conf
Retrieving file: /boot/extlinux/extlinux.conf
1:      default
Retrieving file: /boot/extlinux/../uInitrd
Retrieving file: /boot/extlinux/../Image
append: earlyprintk console=tty1 console=ttyS2,115200n8 console=both rw root=PARTUUID=83e64bff-f2f6-4d3f-b74a-0d7282ba02db rootwait rootfstype=ext4 fsck.repair=yes loglt
Retrieving file: /boot/extlinux/../rockchip/rk3568-nanopi-r5s.dtb
Moving Image from 0x2080000 to 0x2200000, end=3ea0000
## Loading init Ramdisk from Legacy Image at 0a200000 ...
   Image Name:   initramfs-6.0.10
   Image Type:   ARM Linux RAMDisk Image (gzip compressed)
   Data Size:    7183123 Bytes = 6.9 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 0a100000
   Booting using the fdt blob at 0xa100000
   Loading Ramdisk to ed80a000, end edee3b13 ... OK
   Loading Device Tree to 00000000ed7eb000, end 00000000ed809d9a ... OK

Starting kernel ...

unable to select a mode : -70
device_remove: Device 'mmc@fe310000.blk' failed to remove, but children are gone
```

closed 2022-11-30 by electrified