pyavitz/debian-image-builder #53: Echo backslash escaped characters

allanmac, 2022-12-08:
Not sure why my bash environment would be unique, but `make help` isn't emitting color/bold characters.

This is fixed by replacing `"@echo "..."` with `"@echo -e "..."`.

pyavitz, 2022-12-08:
I just get an `-e`

```sh
odroidn2plus: debian-image-builder git:( feature U:1 ?:19 ✗ ) $ make help

-e 			Debian Image Builder
			********************
Boards:
  Allwinner:  nanopim1 nanopineo nanopi opione opipc opir1 pine64 tritium
  Amlogic:    lepotato odroidc4/hc4 odroidn2/plus radxazero
  Broadcom:   raspizero raspi2 raspi3 raspi4
  Rockchip:   nanopc renegade rock64 rockpro64
  Samsung:    odroidxu4
  WIP:        cubietruck cuboxi bananapim5 bananapip2zero nanopir4s/e/r5s
              odroidc1/m1 pinebookpro
```

allanmac, 2022-12-08:
I'm running `GNU bash, version 5.2.1(1)-release`.  

There might be other differences like how color support is enabled in our environments.

pyavitz, 2022-12-08:
```sh
patrick@renegade:~$ echo "${BASH_VERSION}"
5.2.2(1)-release
patrick@renegade:~$ cat /etc/os-release
PRETTY_NAME="Ubuntu 22.10"
NAME="Ubuntu"
VERSION_ID="22.10"
VERSION="22.10 (Kinetic Kudu)"
VERSION_CODENAME=kinetic
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=kinetic
LOGO=ubuntu-logo
```

Are you on Ubuntu Kinetic Kudu or Bookworm? The builder doesn't currently support those, as one is a short term release and the other a current unstable. As much as I would like too, it's to difficult to support such releases.

allanmac, 2022-12-08:
I'm on a Debian Bullseye variant.

pyavitz, 2022-12-08:
Did you manually update to this version of bash? My bullseye installs are @ 5.1.4(1)-release.

allanmac, 2022-12-08:
It's a company modified version of Debian so I'll close this issue and try again on stock Bullseye. 
Thanks!

closed 2022-12-08 by allanmac