pyavitz/debian-image-builder #61: Uboot Packaging

ShorTie8, 2023-01-09:
In my Gentoo version I set a BUILD_VERSION which messes up uboot.
but with BV="${UBOOT_VERSION}", all is well.
Think it might be better too...

`diff --git a/packaging/pkgvars b/packaging/pkgvars
index 58030726..72ecd43f 100644
--- a/packaging/pkgvars
+++ b/packaging/pkgvars
@@ -23,9 +23,10 @@ echo compression='"'$COMPRESSION'"' >> "scripts/package/pkgvars"
 }
 
 uboot_packaging(){
+echo -e "${PNK}[${GRN}Uboot Packaging BUILD_VERSION ${UBOOT_VERSION} ${PNK}]${FIN}"
 PKGNAME="u-boot-${SERIES}"
 PKGVERSION=`echo $UBOOT_VERSION | sed 's/v//g'`
-BV="${BUILD_VERSION}"
+BV="${UBOOT_VERSION}"
 mkdir -p ${PKGNAME}_${PKGVERSION}-${BV}_${ARCH_EXT}/DEBIAN
 mkdir -p ${PKGNAME}_${PKGVERSION}-${BV}_${ARCH_EXT}/usr/lib/${PKGNAME}/
 if [[ "$FAMILY" == "allwinner" ]]; then`

pyavitz, 2023-01-09:
That's because `BUILD_VERSION=` is already in use in the userdata.txt file.

example:
```sh
u-boot-exynos5422_2022.10-3_armhf.deb
```

usage:
```sh
u-boot-exynos5422_2022.10-'BUILD_VERSION="3"'_armhf.deb
```

It works for both uboot and linux packages.

ShorTie8, 2023-01-09:
Ended up it was blank is why.
My bad

closed 2023-01-09 by ShorTie8