pyavitz/debian-image-builder #63: USB Drive support

ShorTie8, 2023-04-24:
This patch allows you to install to a usb device using inotify-wait.
So it ends up being like a 6k patch and a Find_USB script.
make usb board=rock5b, calls Find_USB and writes the drive to USB_Drive.txt.
USB_Drive.txt is used to determine whether to use losetup or not.

USB.diff
`diff --git a/Makefile b/Makefile
index 25b2e9ea..4841a202 100644
--- a/Makefile
+++ b/Makefile
@@ -170,6 +170,17 @@ usbboot:
 	# Creating image
 	$(call build_usbboot)
 
+usb: 
+	# Creating USB Device
+	@bash ./scripts/Find_USB
+	# Compiling u-boot
+	$(call build_uboot)
+	# Compiling kernel
+	$(call build_kernel)
+	# Creating ROOTFS tarball
+	$(call create_rootfs)
+	# Creating image
+	$(call build_image)
 # MISCELLANEOUS
 menu: 
 	# Menu
diff --git a/install.sh b/install.sh
index a5dac173..59b21e3a 100755
--- a/install.sh
+++ b/install.sh
@@ -83,6 +83,7 @@ if [[ "$ARCH" == "aarch64" ]]; then
 		make ncompile
 	fi
 fi
+apt install -y eject inotify-tools
 
 # install builder theme
 make dialogrc
diff --git a/lib/function/staging b/lib/function/staging
index 68ddb70f..739be438 100644
--- a/lib/function/staging
+++ b/lib/function/staging
@@ -26,7 +26,18 @@ fi
 }
 
 single_partition(){
-truncate -s ${IMGSIZE} "${IMAGE_FOLDER}${IMAGE_FILE_NAME}"
+if [[ -f USB_Drive.txt ]]; then
+	USB_Drive="$(cat USB_Drive.txt)"
+	IMAGE_LOOP_DEV="${USB_Drive}"
+	Disk="${USB_Drive}"
+	Size=""
+	dd if=/dev/zero of=${USB_Drive} bs=4K count=420 conv=notrunc status=progress
+else
+	truncate -s ${IMGSIZE} "${IMAGE_FOLDER}${IMAGE_FILE_NAME}"
+	IMAGE_LOOP_DEV="$(losetup --show -P -f ${IMAGE_FOLDER}${IMAGE_FILE_NAME})"
+	Disk="${IMAGE_FOLDER}${IMAGE_FILE_NAME}"
+	Size=5999999
+fi
 if [[ "$GPT" == "true" ]]; then
 	if [[ "$FAMILY_EXT" == "rk356x" || "$FAMILY_EXT" == "rk3588" ]]; then
 		sgdisk -og "${IMAGE_FOLDER}${IMAGE_FILE_NAME}"
@@ -46,13 +57,16 @@ else
 	a
 	t
 	b
-	5999999
+	${Size}
 
 	p
-	w" | fdisk "${IMAGE_FOLDER}${IMAGE_FILE_NAME}"
+	w" | fdisk "${Disk}"
+fi
+if [[ -f USB_Drive.txt ]]; then
+	IMAGE_LOOP_DEV_ROOTFS="${USB_Drive}1"
+else
+	IMAGE_LOOP_DEV_ROOTFS="${IMAGE_LOOP_DEV}${P_ROOTFS}"
 fi
-IMAGE_LOOP_DEV="$(losetup --show -P -f ${IMAGE_FOLDER}${IMAGE_FILE_NAME})"
-IMAGE_LOOP_DEV_ROOTFS="${IMAGE_LOOP_DEV}${P_ROOTFS}"
 partprobe "${IMAGE_LOOP_DEV}"
 mkdir -p ${P_ROOTFS}
 filesystem
@@ -68,15 +82,27 @@ $MOUNT_ROOTFS
 }
 
 dual_partition(){
-truncate -s ${IMGSIZE} "${IMAGE_FOLDER}${IMAGE_FILE_NAME}";
+if [[ -f USB_Drive.txt ]]; then
+	USB_Drive="$(cat USB_Drive.txt)"
+	IMAGE_LOOP_DEV="${USB_Drive}"
+	Disk="${USB_Drive}"
+	Size=" "
+	dd if=/dev/zero of=${USB_Drive} bs=1M count=420 conv=notrunc status=progress
+else
+	truncate -s ${IMGSIZE} "${IMAGE_FOLDER}${IMAGE_FILE_NAME}";
+	IMAGE_LOOP_DEV="$(losetup --show -P -f ${IMAGE_FOLDER}${IMAGE_FILE_NAME})"
+	Disk="${IMAGE_FOLDER}${IMAGE_FILE_NAME}"
+	Size="+3G"
+	IMAGE_LOOP_DEV_BOOT="${IMAGE_LOOP_DEV}${P_BOOT}"
+fi
 if [[ "$VFAT" == "true" ]]; then
 	if [[ "$GPT" == "true" ]]; then
-		parted --script "${IMAGE_FOLDER}${IMAGE_FILE_NAME}" \
+		parted --script "${Disk}" \
 		mklabel gpt \
 		mkpart 1 fat32 16MiB 256MiB \
 		mkpart 2 ext2 256MiB 100%
 		if [[ "$EFI" == "true" ]]; then
-			parted --script "${IMAGE_FOLDER}${IMAGE_FILE_NAME}" set 1 esp on
+			parted --script "${Disk}" set 1 esp on
 		fi
 	else
 		echo -e "o
@@ -93,14 +119,14 @@ if [[ "$VFAT" == "true" ]]; then
 		p
 		2
 		526336
-
+		
 		p
-		w" | fdisk "${IMAGE_FOLDER}${IMAGE_FILE_NAME}";
+		w" | fdisk "${Disk}";
 		if [[ "$EFI" == "true" ]]; then
 			if [[ "$FAMILY" == "broadcom" ]]; then
 				:;
 			else
-				parted --script "${IMAGE_FOLDER}${IMAGE_FILE_NAME}" set 1 esp on
+				parted --script "${Disk}" set 1 esp on
 			fi
 		fi
 	fi
@@ -115,13 +141,17 @@ else
 	p
 	2
 	526336
-
+	
 	p
-	w" | fdisk "${IMAGE_FOLDER}${IMAGE_FILE_NAME}";
+	w" | fdisk "${Disk}";
+fi
+if [[ -f USB_Drive.txt ]]; then
+	IMAGE_LOOP_DEV_BOOT="${USB_Drive}1"
+	IMAGE_LOOP_DEV_ROOTFS="${USB_Drive}2"
+else
+	IMAGE_LOOP_DEV_BOOT="${IMAGE_LOOP_DEV}${P_BOOT}"
+	IMAGE_LOOP_DEV_ROOTFS="${IMAGE_LOOP_DEV}${P_ROOTFS}"
 fi
-IMAGE_LOOP_DEV="$(losetup --show -P -f ${IMAGE_FOLDER}${IMAGE_FILE_NAME})"
-IMAGE_LOOP_DEV_BOOT="${IMAGE_LOOP_DEV}${P_BOOT}"
-IMAGE_LOOP_DEV_ROOTFS="${IMAGE_LOOP_DEV}${P_ROOTFS}"
 partprobe "${IMAGE_LOOP_DEV}"
 mkdir -p ${P_BOOT} ${P_ROOTFS}
 filesystem
diff --git a/scripts/stage1 b/scripts/stage1
index 20953e8a..2bb35e43 100755
--- a/scripts/stage1
+++ b/scripts/stage1
@@ -158,7 +158,14 @@ if [[ "$P_VALUE" == "p1" ]]; then
 		resize2fs -M "${IMAGE_LOOP_DEV_ROOTFS}"
 		tune2fs -Ohas_journal "${IMAGE_LOOP_DEV_ROOTFS}"
 	fi
-	losetup -d "${IMAGE_LOOP_DEV}";
+	if [[ -f USB_Drive.txt ]]; then
+		echo -e "${PNK}[${GRN}Ejecting USB $USB_Drive ${YLW}$P_VALUE ${GRN}Install${PNK}]${FIN}"
+		eject -v $USB_Drive
+		rm -vf  {board.txt,kernel.txt,part-uuid.txt,USB_Drive.txt}
+		exit 0
+	else
+		losetup -d "${IMAGE_LOOP_DEV}";
+	fi
 	mv "${IMAGE_FOLDER}${IMAGE_FILE_NAME}" "${IMAGE_FILE_NAME}";
 	if [ $RAM -ne 0 ]; then
 		umount "${IMAGE_FOLDER}";
@@ -252,7 +259,14 @@ if [[ "$P_VALUE" == "p2" ]]; then
 		resize2fs -M "${IMAGE_LOOP_DEV_ROOTFS}"
 		tune2fs -Ohas_journal "${IMAGE_LOOP_DEV_ROOTFS}"
 	fi
-	losetup -d "${IMAGE_LOOP_DEV}";
+	if [[ -f USB_Drive.txt ]]; then
+		echo -e "${PNK}[${GRN}Ejecting USB $USB_Drive ${YLW}$P_VALUE ${GRN}Install${PNK}]${FIN}"
+		eject -v $USB_Drive
+		rm -vf  {board.txt,kernel.txt,part-uuid.txt,USB_Drive.txt}
+		exit 0
+	else
+		losetup -d "${IMAGE_LOOP_DEV}";
+	fi
 	mv "${IMAGE_FOLDER}${IMAGE_FILE_NAME}" "${IMAGE_FILE_NAME}";
 	if [ $RAM -ne 0 ]; then
 		umount "${IMAGE_FOLDER}";
`

Find_USB
`#! /bin/bash
# COLOR
RED="\e[0;31m"
GRN="\e[0;32m"
BLU="\e[0;34m"
PNK="\e[0;35m"
YLW="\e[0;36m"
WHT="\033[0m"
FIN="\e[0m"

case `basename $0` in
  Find_USB)
    # Won't work as a mere mortal
    if [ $UID -ne 0 ]; then
      echo "This script MUST be run as root. Please su or sudo and try again!"
      echo
      exit 1
    fi
    ;;
esac

echo -e "\n\n${RED}[${WHT}[${BLU}[${GRN}  Pleaze Plug in ${YLW}USB${GRN} Device  ${BLU}]${WHT}]${RED}]${FIN}"
# Wait until a new [0-9] node appears in /dev
#   This should be OK because there should be few device changes during operation
DEV_FILE="/tmp/inotify_devs"
INOTIFY_CMD="inotifywait -q"
WATCH="CREATE"

# Watch for CREATEs in /dev of xxx[0-9]*
$INOTIFY_CMD -m --exclude "t[my]" /dev | while read a b c; do
  if [ "$b" != "$WATCH" ]; then continue; fi
  C="${c/[0-9]*/}"
  if [ "$c" != "$C" -a -e "$a$C" -a -e "$a${C}1" ]; then
    echo "${C}" > ${DEV_FILE}
    exit  # This exits the 'while read' subprocess
  fi
done
read fldev <$DEV_FILE
rm -f $DEV_FILE

echo; echo "Flash drive is /dev/${fldev}"; echo
echo "/dev/${fldev}" > USB_Drive.txt

MounteD=`cat /etc/mtab | grep $fldev`
echo -e "\n${PNK}[${GRN}      Flash drive is $fldev ${PNK}]${FIN}"
if [ "${MounteD}" != "" ]; then
	echo "  Un Mounting $fldev "
	for i in {1..9}; do
		NAME=`cat /etc/mtab | grep "$fldev$i" | cut -d " " -f2`
		if [ "$NAME" != "" ]; then
			umount $NAME
		fi
      done
else
	echo "  $fldev is not mounted, continuing "
fi

exit 0
`

pyavitz, 2023-04-25:
What exactly am I looking at here? Is it writing an img created by the builder, to USB, from the builder?

ShorTie8, 2023-04-25:
Instead of creating a image, writing to a loopback, it uses /dev/sdX, a usb device.

pyavitz, 2023-04-26:
Well when you and github become friends create a PR.

ShorTie8, 2023-05-01:
I reversed my comment and it killed the pr, lol.
So we start out adding in USB_Drive into custom.txt bye,
`diff --git a/lib/dialog/config b/lib/dialog/config
index 5c9df0e3..73838641 100755
--- a/lib/dialog/config
+++ b/lib/dialog/config
@@ -213,6 +213,9 @@ echo "" >> custom.txt
 echo "# Compression Types: xz zst" >> custom.txt
 echo 'IMG_COMPRESSION="xz"' >> custom.txt
 echo "" >> custom.txt
+echo "# USB  Drive: sdX" >> custom.txt
+echo 'USB_Drive=""' >> custom.txt
+echo "" >> custom.txt
 echo "### DO NOT EDIT BELOW THIS LINE" >> custom.txt
 echo "# CUSTOM=${BUILDER}" >> custom.txt
 }
`

??

ShorTie8, 2023-05-14:
Gona do up a more generic Dev_Device for all

closed 2023-05-14 by ShorTie8