pyavitz/debian-image-builder #66: bdone

ShorTie8, 2023-05-14:
It does not really tell you it finished
So I say, bdone you thing, lol.
[bdone.patch](https://github.com/pyavitz/debian-image-builder/files/11471126/bdone.patch)

pyavitz, 2023-05-14:
I've been slowly removing those echoes from the builder. I'm just to lazy to go through the function file and see what is still in use. As for knowing when the image is done, it's pretty obvious when it's finished being created I thought? Everything stops :) If using the menu interface it actually shows you the results.

I added your shrink idea, but placed it in the custom.txt file. Less effort. https://github.com/pyavitz/debian-image-builder/commit/52f4103eb2067f04f7ab5338484f0e87e54b131f

ShorTie8, 2023-05-14:
Ya, noticed some not used at all and some only once.
Let me weed thru it and see what i get.

ShorTie8, 2023-05-14:
Try this maybe ??
Played colors to .. :/~
[echoes.patch](https://github.com/pyavitz/debian-image-builder/files/11472968/echoes.patch)

pyavitz, 2023-05-17:
ECHOES are the least of this builders problems right now. I'm closing this, you can re-open it at a later date if for whatever reason it doesn't sort its self out.

closed 2023-05-17 by pyavitz