pyavitz/debian-image-builder #68: shrink errors

ShorTie8, 2023-05-20:
Don't know why, but shrink error's out on me at times .. (~
kpart2 is empty for some reason.
Looking around I found PiShrink on github.
So, We add it as a submodule with
`git submodule add https://github.com/Drewsif/PiShrink.git scripts/PiShrink`
And apply the patch
[pishrink.patch](https://github.com/pyavitz/debian-image-builder/files/11522066/pishrink.patch)

pyavitz, 2023-05-20:
I myself haven't been having any issues with it. The reason I stopped using the PiShrink script is it wasn't shrinking the images correctly and in some cases `arm` would shrink the image to much. If for whatever reason you received an error at some point it is possible that kpartx didn't unmount the loop?

When hitting image errors, its good practice to run `make cleanup` before trying to build the image again.

Could also disable shrinking in the custom.txt file?

ShorTie8, 2023-05-20:
You ever think "it wasn't shrinking the images correctly" twas a bug in it they fixed ??

pyavitz, 2023-05-21:
Not really, as I moved away from using that script some time ago. I ended up using a fork of it eventually and then further cutting into the fork. I haven't looked at it in a while, but does it even use kpartx? The kpartx bit in the builder is used to mount the img, to update the PARTUUID after the shrink.

ShorTie8, 2023-05-21:
Nop, No kpartx now
That is what is giving me problems.

pyavitz, 2023-05-21:
This is the only piece of the script I use anymore: https://github.com/pyavitz/debian-image-builder/blob/feature/scripts/shrink

You probs had a kpartx issue and it left open loops in /dev/mapper. Just a guess, but I can't think of any other reason it would be creating a problem?

ShorTie8, 2023-05-22:
Oh really ??
Then why is it that kpartx is what keeps failing then ??
It does not populate kpart2 for me

pyavitz, 2023-05-22:
What host OS are you using and ARCH. Can you mount an IMG using kpartx?

`kpartx -av filename.img`

ShorTie8, 2023-05-23:
arm64 Debian 11.7
Most like not on the log files, Sorry...

ShorTie8, 2023-05-27:
Don't like none of my help I guess.
Oh well.

closed 2023-05-27 by ShorTie8

pyavitz, 2023-05-30:
You didn't help. You suggested I use pishrink, but that has nothing to do kpartx.

ShorTie8, 2023-05-30:
Well, maybe if you had given me more help in the channel when i twas asking/talking about it.I mean, like all I got ~"it works fine for me!!"Maybe PiShrink has figured out that kpartx is NOT the best and moved on to a newer better way ??That is why thier images work in Windows and Yours do not !!But I know, You Just Don't Care !!

pyavitz, 2023-05-31:
Well you are right, I don't really care what happens in Windows. Unless I am missing something, pishrink has never used kpartx, it is something I decided to use in the builder after originally using that script in order to make sure the PARTUUID could be found and updated in the images.

There is some confusion here and I believe you are getting upset over nothing. If you can think of a better way to shrink ext4 based images and still retain or update the PARTUUID afterwards, without the use of kpartx I am all ears.

ShorTie8, 2023-05-31:
PARTUUID afterwards
That is what is most likely messing things up .. :/~Why, If I can ask ??

pyavitz, 2023-05-31:
I believe it has something to do with `resize2fs`, although it could be a combination of things. Either way, the same thing happens when growing a filesystem, especially on `ARCH=arm`. The `PARTUUID` can also change when flashing u-boot bins, but I find that happens more when using an SDCARD and not so much on eMMC.

Either way it happens, so I had to account for it in the builder and other scripts.

ShorTie8, 2023-06-01:
Ok, I'll revert back and play more.See I did this shrinky thing myself 8 years ago.ShorTie8/Image_Shrinker: A simple script to shrink a image of unwanted free space (github.com)
I just used a little math for it.But got me.I'll logging on and go back to the channel.Ok ??Later