pyavitz/debian-image-builder #81: Support for Youyeetoo YY3568

javentous, 2023-11-04:
Hello and thanks for your work,

Would it be possible to add support to build an image for the vendor Youyeetoo and their RK3568 based board, YY3568?

Here is a link to their [wiki entry for YY3568](http://wiki.youyeetoo.com/en/YY3568), and some quick links below:
- [Google Drive](https://drive.google.com/drive/folders/1Nt3hKzRbzeD6CnMvHa4dG4xxvH-jomkq) link with all their files
- Link to their [Debian images](https://drive.google.com/drive/folders/1RcNVCGJCfITypKraSi_ppmYWpQKvDl_R)
- They have another folder for [Debian images with docker support](https://drive.google.com/drive/folders/13DnF6r-4XVy2gLJ4VKGUZXvTY_X6_1m6)

Thanks again,
Jawad

pyavitz, 2023-11-04:
These people need a Github. I would need to download 4GB just to checkout the source code. I would guess this could be mainlined as the SoC is fairly well supported now. But not having the unit myself and the source code being buried inside a google drive, I personally wouldn't make an attempt.

javentous, 2023-11-04:
Ah, noted. Appreciate the honesty.

Lindorak, 2023-11-06:
Sounds like a great addition. Hopefully this will be implemented soon.

javentous, 2023-11-13:
It seems like they have a github repo, I think they'll add the files soon enough.

https://github.com/youyeetoo/YY3568

closed 2024-05-02 by pyavitz