pyavitz/debian-image-builder #87: USB lacking

wdthompson, 2023-12-11:
After several tries I managed to frankenstien an image (orangepi5)
There seems to be a problem with USB lsusb can show 4 hubs or 6 or 8,, 8 is correct
The vertical usb2 is often the 1st  to go missing
There is some ? intelegent ? discussion
https://github.com/armbian/build/pull/4582

pyavitz, 2023-12-11:
https://github.com/armbian/build/pull/4634

```
sudo su
tee /etc/systemd/system/orangepi5-usb2-init.service <<EOF
[Unit]
Description=Init USB2 for Orange Pi 5

[Service]
ExecStart=/usr/bin/sh -c "echo host > /sys/kernel/debug/usb/fc000000.usb/mode"
Type=oneshot

[Install]
WantedBy=default.target
EOF
systemctl enable orangepi5-usb2-init.service
```
reboot and let me know if it works.

wdthompson, 2023-12-16:
OK, I recompiled and installed that kernel, a default kernel, uname-r  5.10.110
There is nothing in /sys/kernel/debug, an empty directory
When both of the horizontal usb ports are used, the usb2 port sometimes locks
Also,C&P,,, I use serial terminal ALL the time, I guess this is why it doesn't work
ls -l /dev/ttyS*
crw-rw---- 1 root dialout 4, 73 Dec 16 02:07 /dev/ttyS9
comparing (kernel) config with armbian (5.10.160), based on ubuntu, except for peripherals, differences
RT[2500USB,73USB,2800USB],, all =m  ,, I don't know if this is significant

pyavitz, 2023-12-16:
> OK, I recompiled and installed that kernel, a default kernel, uname-r 5.10.110 There is nothing in /sys/kernel/debug, an empty directory When both of the horizontal usb ports are used, the usb2 port sometimes locks Also,C&P,,, I use serial terminal ALL the time, I guess this is why it doesn't work ls -l /dev/ttyS* crw-rw---- 1 root dialout 4, 73 Dec 16 02:07 /dev/ttyS9 (works for uboot, has NEVER worked after kernel starts loading) comparing (kernel) config with armbian (5.10.160), based on ubuntu, except for peripherals, differences RT[2500USB,73USB,2800USB],, all =m ,, I don't know if this is significant I can say that both the usb3 ports work, maybe not simultaneously though,, because lsusb Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 003 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub There should be 2 - 3.0hubs and 4 - 2.0hubs As well, suspend seems to lock about 1/2 the time sometimes a pwr button press works, sometimes not,, but that doesn't really matter since it is hooked to line (I set it to lock screen)

I don't own this unit or any RK35XX. It was added by request, so anything that does or does not work properly, I have no clue. I can setup the board to pull the Armbian kernel source if you like and see if that makes a difference?

wdthompson, 2023-12-18:
I think, maybe, that it is a dtb problem
Oh I love hacking dtb's  -- NOT
doing the diff's on various dts's, nothing is jumping out at me,
regarding the 6 hubs vrs 8
I still have a couple more to look at, what fun

pyavitz, 2023-12-18:
> I think, maybe, that it is a dtb problem
Oh I love hacking dtb's -- NOT
doing the diff's on various dts's, nothing is jumping out at me,
regarding the 6 hubs vrs 8
I still have a couple more to look at, what fun
One small win,, 6.7.0-rc3-4-MANJARO-ARM-RK3588 now has 6 hubs (previous 4, no usbC still)
doing a diff on something so different is pretty much pointless

What Armbian is doing in that service file is changing it to host mode.
```
ExecStart=/usr/bin/sh -c "echo host > /sys/kernel/debug/usb/fc000000.usb/mode"
```

I haven't looked at the DTS, but my guess would be it is set to OTG.

Which version of the OrangePi 5 are you using?

pyavitz, 2023-12-18:
I could create a patch for it and switch it to host, but I have no clue if that would fix the problem.

https://github.com/orangepi-xunlong/linux-orangepi/blob/orange-pi-5.10-rk3588/arch/arm64/boot/dts/rockchip/rk3588s-orangepi-5.dtsi#L358

https://github.com/orangepi-xunlong/linux-orangepi/blob/orange-pi-5.10-rk3588/arch/arm64/boot/dts/rockchip/rk3588s-orangepi.dtsi#L546

I would think the service file I listed above would be enough to test if that works or not.

wdthompson, 2023-12-19:
I think it might be worth trying the patch, this is a quite confusing issue
Probably because the usbC has 3 modes, normal, otg , normal and alt-mode for video
I have a opi5, not opi5b or opi5+
opi5 has 4 usb ports,  3 are type A, top horizontal is usb3
Somehow the vertical is linked to usbC, for rkdeveloptool (or gui) vertical MUST be empty
I've been a little lazy, still have to diff armbian and rebornOS (actually dreading)
It is most confusing when so complex
The armbian service is for a differently configured kernel, for me /sys/kernel/debug is an empty directory
I did some searching in /sys/kernel, didn't find any appropriate place

pyavitz, 2023-12-19:
`git pull` and rebuild. Let me know how it goes.

wdthompson, 2023-12-21:
And it didn't work, 3 times (maybe I heated cpu 4X?)
All this (compile) on opi5-jammy, all ended (serial-uboot) Starting kernel and on
screen (only display shown) Loading, please wait
Serial has (I have never seen)   Error: Invalid FAT entry: 0x00000000  
immediatly after initrd, kernel, and dtb load
Also, Image...Load Address: 00000000
I am not sure this is right, I seem to recall something different in cortex days, but that is 10ya
I made a new directory and a fresh git clone, changed as little as possible,
for this the image creation was ...difficult. Under the menu system couldn't find image, but it was there... img/
manually, the tests in stage1 (line23) maybe it is the [test] vrs [[test]] problem again,
I plan to test that tonight (bypassed USERACCT 1)

wdthompson, 2023-12-21:
Sorry, line 22 (stage1), somehow the quotes are not working, if NAME has a space, as most names do,
then [ -z $NAME ] chokes
This new attempt with a working make image fails exactly the same way

pyavitz, 2023-12-21:
Try these; https://github.com/pyavitz/binary/releases/tag/images

Observe the USAGE section to create a username and password during boot. Let me know which one if any work.

wdthompson, 2023-12-22:
408 gave same results as above, well 1 more line, starting systemd-udev
518 works OK, 8 usb hubs, serial works (ttyFIQ0 whatever that is)
adduser (at 1st boot) did not work, I had to edit /etc/shadow to make a root boot, then adduser in that terminal
OR did you mean to add userdata.txt somewhere into the image?
OR did you mean when using the source?

pyavitz, 2023-12-22:
> 408 gave same results as above, well 1 more line, starting systemd-udev 518 works OK, 8 usb hubs, serial works (ttyFIQ0 whatever that is) adduser (at 1st boot) did not work, I had to edit [...]

No, the /boot/useraccount.txt file. You can't have any WHITE SPACES or ODD CHARACTERS in the NAME, USERNAME or PASSWORD variables.

So it boots now and the USB ports work? Good.

closed 2024-01-15 by pyavitz