pyavitz/debian-image-builder #88: scripts/shrink problem!!!

laooopooo, 2023-12-15:
scripts/shrink: 行 83: 676832 - 预计文件系统的最小尺寸：676832: 语法错误：需要操作数（错误记号是 "预计文件系统的最小尺寸：676832"）
scripts/shrink: 行 93: 预计文件系统的最小尺寸：676832 * 4096: 语法错误：需要操作数（错误记号是 "预计文件系统的最小尺寸：676832 * 4096"）
scripts/shrink: 行 94: 4194304 + : 语法错误：需要操作数（错误记号是 "+ "）
Shrunk meson-g12a-radxa-zero-debian-bookworm-2023-12-15.img from 3.9G to 16K

pyavitz, 2023-12-15:
Odd. Kind of looks like the IMG was shrunk twice?

For the time being disable shrink; `sed -i "s/^ENABLE_SHRINK=.*/ENABLE_SHRINK="'"false"'"/" custom.txt`

pyavitz, 2023-12-15:
What host machine and distro are you using?

HOST: arm64
DISTRO: bookworm

```
+ [[ true == \t\r\u\e ]]
+ [[ false == \f\a\l\s\e ]]
+ [[ ext4 == \e\x\t\4 ]]
+ tune2fs '-O^has_journal' /dev/loop0p1
tune2fs 1.47.0 (5-Feb-2023)
+ fsck -fy /dev/loop0p1
fsck from util-linux 2.38.1
e2fsck 1.47.0 (5-Feb-2023)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
ROOTFS: 63789/249984 files (0.1% non-contiguous), 457368/998912 blocks
+ resize2fs -M /dev/loop0p1
resize2fs 1.47.0 (5-Feb-2023)
Resizing the filesystem on /dev/loop0p1 to 619892 (4k) blocks.
The filesystem on /dev/loop0p1 is now 619892 (4k) blocks long.

+ tune2fs -Ohas_journal /dev/loop0p1
tune2fs 1.47.0 (5-Feb-2023)
Creating journal inode: done
+ losetup -d /dev/loop0
+ mv img/meson-g12a-radxa-zero-debian-bookworm-2023-12-15.img meson-g12a-radxa-zero-debian-bookworm-2023-12-15.img
+ '[' 0 -ne 0 ']'
+ rmdir img/
+ rmdir p1
+ [[ true == \t\r\u\e ]]
+ [[ false == \f\a\l\s\e ]]
+ [[ ext4 == \e\x\t\4 ]]
+ shrink
++ cat /etc/os-release
++ grep -w VERSION_CODENAME
++ sed s/VERSION_CODENAME=//g
+ RELEASE=bookworm
+ [[ -f scripts/shrink ]]
+ chmod +x scripts/shrink
+ scripts/shrink -s meson-g12a-radxa-zero-debian-bookworm-2023-12-15.img
ROOTFS: 63789/153216 files (0.3% non-contiguous), 466710/619892 blocks
resize2fs 1.47.0 (5-Feb-2023)
Shrunk meson-g12a-radxa-zero-debian-bookworm-2023-12-15.img from 3.9G to 2.4G
+ [[ p1 == \p\1 ]]
+ kpartx_p1
+ [[ p1 == \p\1 ]]
+ echo 'ROOTFS_LOOP="/dev/mapper/'
+ echo '"'
+ IMAGE_FILE=meson-g12a-radxa-zero-debian-bookworm-2023-12-15.img
++ kpartx -av meson-g12a-radxa-zero-debian-bookworm-2023-12-15.img
+ KPARTX='add map loop0p1 (253:0): 0 4959137 linear 7:0 8192'
+ echo 'add map loop0p1 (253:0): 0 4959137 linear 7:0 8192'
add map loop0p1 (253:0): 0 4959137 linear 7:0 8192
+ [[ bookworm == \j\a\m\m\y ]]
+ grep -o loop.p.
+ paste -d '\0' kpart1 kpart2 kpart3
+ rm -f kpart1 kpart2 kpart3
+ source kpart.txt
++ ROOTFS_LOOP=/dev/mapper/loop0p1
+ ROOTFS=/dev/mapper/loop0p1
++ blkid -o export -- /dev/mapper/loop0p1
++ sed -ne 's/^PARTUUID=//p'
+ ROOT_PARTUUID=420f68cc-01
+ mkdir -p p1
+ sleep 1s
+ mount /dev/mapper/loop0p1 p1
+ echo ''

+ [[ true == \t\r\u\e ]]
+ [[ false == \f\a\l\s\e ]]
+ extlinux
+ [[ false == \t\r\u\e ]]
+ [[ arm64 == \a\r\m ]]
+ sleep 3s
+ umount p1
+ rm -fdr p1
+ kpartx -d meson-g12a-radxa-zero-debian-bookworm-2023-12-15.img
+ rm -f kpart.txt
+ [[ p1 == \p\2 ]]
+ finish_image
++ date +%T
++ sed 's/..$//'
++ sed s/://g
+ IMAGE_TIME=0735
+ IMAGE_FIN_NAME=meson-g12a-radxa-zero-debian-bookworm-6.1.68-arm64-ext4-2023-12-15-0735.img
+ [[ -f meson-g12a-radxa-zero-debian-bookworm-2023-12-15.img ]]
+ mv -f meson-g12a-radxa-zero-debian-bookworm-2023-12-15.img meson-g12a-radxa-zero-debian-bookworm-6.1.68-arm64-ext4-2023-12-15-0735.img
+ '[' 0 -eq 1 ']'
+ rm -f part-uuid.txt board.txt kernel.release
+ exit 0
```

Not seeing this error.

laooopooo, 2023-12-18:
my host is:
HOST ARCH: x86_64
HOST DISTRO: ubuntu 22.04

closed 2023-12-18 by pyavitz