pyavitz/debian-image-builder #89: devuan-->testing version,progress not passed

laooopooo, 2023-12-18:
configs:
### DISTRIBUTION AND RELEASE
DISTRO="devuan"
DISTRO_VERSION="testing"

(Reading database ... 4985 files and directories currently installed.)
Preparing to unpack .../archives/adduser_3.137_all.deb ...
Unpacking adduser (3.137) ...
Preparing to unpack .../cron-daemon-common_3.0pl1-180_all.deb ...
Unpacking cron-daemon-common (3.0pl1-180) ...
Setting up adduser (3.137) ...
dpkg: dependency problems prevent configuration of cron-daemon-common:
 cron-daemon-common depends on systemd | systemd-standalone-sysusers | systemd-sysusers; however:
  Package systemd is not installed.
  Package systemd-standalone-sysusers is not installed.
  Package systemd-sysusers is not installed.

dpkg: error processing package cron-daemon-common (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 cron-daemon-common

pyavitz, 2023-12-18:
This happens using host?
```
HOST ARCH: x86_64
HOST DISTRO: ubuntu 22.04
```

pyavitz, 2024-05-02:
https://www.devuan.org/os/announce/excalibur-usrmerge-announce-2024-02-20.html
https://dev1galaxy.org/viewtopic.php?id=6340

pyavitz, 2024-05-03:
Using Ubuntu Jammy and `mmdebstrap` I was able to get two rootfs tarballs made.
https://github.com/pyavitz/rootfs-tarballs/releases/tag/rootfs

Not sure why it only works using Jammy and not Bookworm or Noble? But it's something.
```
   Devuan GNU/Linux 6 (excalibur/ceres) 6.6.30 #1 Fri May 3 05:49:11 EDT 2024
                                    aarch64

                                                       Hardkernel ODROID-N2Plus

── DISK
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p1   30G  2.1G   26G   8% /

── NETWORK
Hostname:    devuan
Wired:       eth0 UP XX.0.0.XXX

── SYSTEM
Processors:  Cortex-A53 @ 2016MHz 29°C, Cortex-A73 @ 2400MHz 32°C
Frequencies: 2016MHz 2400MHz
Online:      0-5
Fan:         RPM 0, PWR 0
Governor:    performance
Memory:      3.7G 239M
Entropy:     256
Uptime:      08:46:26 up 0 min, 0 user, load average: 0.14, 0.03, 0.01
```