pyavitz/debian-image-builder #95: g_ether / otg

jayofelony, 2024-02-09:
Hey there!!

First of, love your work. But how do I enable OTG? When I add g_ether to `/etc/modules-load.d/modules.conf` and let it boot, it does not show in ifconfig of my host laptop. Any ideas?

I am using the BPi M4 Zero build.

pyavitz, 2024-02-09:
Because I'm in the middle of debugging things, I ended up changing the `&usbotg` node to `dr_mode = "host"`. Try this overlay and see if it works.

```
wget https://github.com/pyavitz/binary/releases/download/060420/allwinner-usb-otg-mode.dts
add-overlay allwinner-usb-otg-mode.dts
```

Add it to /boot/extlinux/extlinux.conf: ../user-overlays/allwinner-usb-otg-mode.dtbo

Example:
```
fdtoverlays ../user-overlays/allwinner-usb-otg-mode.dtbo
```

Reboot.

jayofelony, 2024-02-09:
Okay, if only I had the means to connect a keyboard haha. I guess I will wait for an update on your end. Thanks for the heads up though!

pyavitz, 2024-02-09:
Give me a sec and I'll adjust the DTS back and push it to git.

jayofelony, 2024-02-09:
Might I also suggest to use pishrink to shrink the final image. That makes it suitable to flash with RPi imager.

pyavitz, 2024-02-09:
> Might I also suggest to use pishrink to shrink the final image. That makes it suitable to flash with RPi imager.

I was using a script based on `pishrink` but its not viable for use with ext4. Which is why I stopped using it. Last I was told, the RPI Imager was working again, after I made changes to the shrink function. Guess not.

Here; partial revert; https://github.com/pyavitz/debian-image-builder/commit/ccd48721b62f49a52719a6437a0de213713e7745

Hopefully the OTG works now.

Sorry about that.

jayofelony, 2024-02-09:
Ok thanks! I will try building a new image immediately. I will keep you posted.

jayofelony, 2024-02-09:
Okay, I can confirm it boots by connecting to a hdmi display. I cannot get any further as I do not have a usb-c keyboard. But on my linux host I do not get an otg connection.

[image]

I added `g_ether` to `/etc/modules-load.d/modules.conf`, also added a `usb0-cfg` to `/etc/network/interfaces.d/`.

pyavitz, 2024-02-09:
> Okay, I can confirm it boots by connecting to a hdmi display. I cannot get any further as I do not have a usb-c keyboard. But on my linux host I do not get an otg connection. [...]

Does this functionality work on the BPI images?

I've only had the unit for like a week and honestly hadn't tested OTG mode until now. I'll investigate and if you find anything let me know.

jayofelony, 2024-02-10:
Okay, it seems to be called cdc_ether. i will try that module

pyavitz, 2024-02-10:
> Okay, it seems to be called cdc_ether. i will try that module
> 
> Edit:
> 
> Also nothing, it does support otg. I know the image supplied by bpi does have otg

Getting there; 
```
── NETWORK
Hostname:    bananapim4zero
Tether:      usb1 UP 10.55.0.1 
Wireless:    wlan0 UP 10.0.0.251 
```

Doesn't seem to broadcast though unless I have a wifi connection going for some reason. Could be my configuration?

jayofelony, 2024-02-10:
You probably know more about this than me, I am just an end user.

pyavitz, 2024-02-10:
> You probably know more about this than me, I am just an end user.

Only time I've ever used OTG was when testing it on Pis. Not honestly all that familiar with the config aspect of it.

jayofelony, 2024-02-10:
Ahhh its called usb1, usually it is called usb0

pyavitz, 2024-02-10:
I don't know. Sometimes it comes up for me and sometimes it doesn't. Could be the PC I'm using... The USB ports on it are kind of iffy.

I'm also using an overlay for it as well and a script I created, but I'm honestly not sure yet if they are required. If you can't get it to come up at all, I'll pass them ur way.

jayofelony, 2024-02-10:
My host ubuntu now sees the device connected, I have a internet sharing script. But there is still no route to host

pyavitz, 2024-02-10:
> My host ubuntu now sees the device connected, I have a internet sharing script. But there is still no route to host.
> 
> SSH is enabled by default?

yes

jayofelony, 2024-02-10:
It works! It is indeed usb1, not usb0 like most the times

jayofelony, 2024-02-10:
[image]

pyavitz, 2024-02-10:
> It works! It is indeed usb1, not usb0 like most the times

good good

jayofelony, 2024-02-10:
`/etc/resolv.conf` doesnt seem to get auto-populated. I manually added `nameserver 8.8.8.8` to make it resolve hostnames

pyavitz, 2024-02-10:
> `/etc/resolv.conf` doesnt seem to get auto-populated. I manually added `nameserver 8.8.8.8` to make it resolve hostnames

As a work-around try;
```
sudo mv /etc/resolv.conf /etc/resolv.conf.orig
echo "nameserver 8.8.8.8" | sudo tee /etc/resolv.conf
echo "nameserver 8.8.4.4" | sudo tee -a /etc/resolv.conf
```
reboot

jayofelony, 2024-02-10:
Yeah, I done that

closed 2024-02-11 by jayofelony