pyavitz/debian-image-builder #96: bluetooth hciuart

jayofelony, 2024-02-11:
[image]
New image new problems haha.

Do you know how to enable bluetooth controller?

pyavitz, 2024-02-11:
it should just work. if its not working than the driver and or firmware didn't load properly. Unfortunately, the builds aren't ready for prime time yet and I'm still working out the bugs. Basically wifi / bt works when it wants to.

```
patrick@bpim4zero:~$ bluetoothctl
Agent registered
[CHG] Controller 98:03:CF:62:3B:40 Pairable: yes
[bluetooth]# show
Controller 98:03:CF:62:3B:40 (public)
	Name: bpim4zero
	Alias: bpim4zero
	Class: 0x00000000
	Powered: yes
	Discoverable: no
	DiscoverableTimeout: 0x000000b4
	Pairable: yes
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0542
	Discovering: no
	Roles: central
	Roles: peripheral
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x05 (5)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
```

pyavitz, 2024-02-29:
With the inclusion of u-boot v2024.01 and a series of fixup patches for rtl8822/21cu this should no longer be an issue.

closed 2024-02-29 by pyavitz