pyavitz/debian-image-builder #97: not loading wifi firmware

jayofelony, 2024-02-11:
[image]
it also seems to not load wifi interface at times

pyavitz, 2024-02-11:
> ![image](https://private-user-images.githubusercontent.com/50277312/303923228-08713b8a-18aa-4e88-9354-2cfd0775f80b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Indeed.

The unit has eMMC issues as well. I've been trying to stabilize the wifi, but have yet to find a solution.

I'm honestly out of ideas on the kernel side and am thinking this might be u-boot related in some way, but I won't know until I get it booting on current mainline.

pyavitz, 2024-02-15:
I've since updated the patching so that the unit now uses the rtl8821cu driver from; https://github.com/morrownr/8821cu-20210916

`git pull` and either upgrade the kernel on the unit or create a new img.

```patrick@bpim4zero:~$ lsmod
Module                  Size  Used by
bnep                   20480  2
zram                   28672  2
8821cu               2281472  0
binfmt_misc            20480  1
btusb                  53248  0
btintel                32768  1 btusb
btrtl                  20480  1 btusb
btbcm                  24576  1 btusb
bluetooth             495616  28 btrtl,btintel,btbcm,bnep,btusb
cfg80211              348160  1 8821cu
ecdh_generic           16384  2 bluetooth
ecc                    28672  1 ecdh_generic
rfkill                 32768  5 bluetooth,cfg80211
snd_soc_sunxi_ahub     20480  2
snd_soc_sunxi_machine    20480  0
sun50iw9_codec         24576  3
pwm_sunxi_enhance      20480  0
snd_soc_sunxi_ahub_dam    16384  2 snd_soc_sunxi_ahub
uio_pdrv_genirq        20480  0
uio                    20480  1 uio_pdrv_genirq
dm_mod                110592  0
dax                    24576  1 dm_mod
```
```
patrick@bpim4zero:~$ bluetoothctl
Agent registered
[CHG] Controller 98:03:CF:62:3B:40 Pairable: yes
[bluetooth]# show
Controller 98:03:CF:62:3B:40 (public)
	Name: bpim4zero
	Alias: bpim4zero
	Class: 0x00000000
	Powered: yes
	Discoverable: no
	DiscoverableTimeout: 0x000000b4
	Pairable: yes
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0542
	Discovering: no
	Roles: central
	Roles: peripheral
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x05 (5)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
```
For reasons unknown it appears to be more stable.

pyavitz, 2024-02-29:
With the inclusion of u-boot v2024.01 and a series of fixup patches for rtl8822/21cu this should no longer be an issue.

closed 2024-02-29 by pyavitz