pyavitz/debian-image-builder #98: bpim4zero: thermal protection

jayofelony, 2024-02-13:
```
Message from syslogd@pwnagotchi at Feb 10 17:37:44 ...
 kernel:thermal thermal_zone0: gpu-thermal: critical temperature reached, shutting down

Message from syslogd@pwnagotchi at Feb 10 17:37:44 ...
 kernel:reboot: HARDWARE PROTECTION shutdown (Temperature too high)
```
I am running it headless over OTG/USB1.

pyavitz, 2024-02-13:
what kernel? 6.1.y vendor?

pyavitz, 2024-02-13:
> ```
> Message from syslogd@pwnagotchi at Feb 10 17:37:44 ...
>  kernel:thermal thermal_zone0: gpu-thermal: critical temperature reached, shutting down
>
> Message from syslogd@pwnagotchi at Feb 10 17:37:44 [...]

Anyway... Not sure what ur doing headless, but It looks to me you over extended and it tripped and shutdown to save its self. Put some heat sinks on it and adjust the scaling.

```
governor -h
CPU frequency scaling
Usage: 

	-c	Conservative
	-o	Ondemand
	-p	Performance
	-s	Schedutil

	-r	Run
	-u	Update
```

jayofelony, 2024-02-13:
6.1 kernel, doing nothing fancy actually. Not using the GPU for sure.

pyavitz, 2024-05-02:
This kernel is hackery and mainline still has a ways to go.

closed 2024-05-02 by pyavitz