pyavitz/debian-image-builder #101: bpi-m2zero

jayofelony, 2024-02-26:
I have succesfully built an image, but I need a `/lib/modules/6.1.31-sun50iw9/build` source folder. Do I have to change a setting in config somewhere?

pyavitz, 2024-02-26:
The headers are already there, you should be able to compile against them.

```
ls /lib/modules/6.1.31/build
arch    Documentation  init      kernel    Module.symvers  scripts   usr
block   drivers        io_uring  lib       net             security  virt
certs   fs             ipc       Makefile  rust            sound
crypto  include        Kconfig   mm        samples         tools
```

pyavitz, 2024-02-26:
The userdata file may see 6.1.79, but currently the kernel source the board is pulling; https://github.com/BPI-SINOVOIP/pi-linux/tree/pi-6.1-sunxi

I'm not sure what you mean, why would an img generated in the builder be looking for `6.1.31-sun50iw9`?

Its only going to use the headers available, unless you are pointing it to look else where?

```
make ARCH=arm64 -j4
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/6.1.31/build M=/home/patrick/8821cu-20210916  modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.31'
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_cmd.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_security.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_debug.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_io.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_ioctl_query.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_ioctl_set.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_ieee80211.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_mlme.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_mlme_ext.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_mi.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_wlan_util.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_vht.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_pwrctrl.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_rf.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_chplan.o
  CC [M]  /home/patrick/8821cu-20210916/core/monitor/rtw_radiotap.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_recv.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_sta_mgt.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_ap.o
  CC [M]  /home/patrick/8821cu-20210916/core/wds/rtw_wds.o
  CC [M]  /home/patrick/8821cu-20210916/core/mesh/rtw_mesh.o
  CC [M]  /home/patrick/8821cu-20210916/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/patrick/8821cu-20210916/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_xmit.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_p2p.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_rson.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_tdls.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_br_ext.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_iol.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_sreset.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_btcoex_wifionly.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_btcoex.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_beamforming.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_odm.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_rm.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_rm_fsm.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_ft.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_wnm.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_mbo.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_rm_util.o
  CC [M]  /home/patrick/8821cu-20210916/core/efuse/rtw_efuse.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_roch.o
  CC [M]  /home/patrick/8821cu-20210916/core/crypto/aes-internal.o
  CC [M]  /home/patrick/8821cu-20210916/core/crypto/aes-internal-enc.o
  CC [M]  /home/patrick/8821cu-20210916/core/crypto/aes-gcm.o
  CC [M]  /home/patrick/8821cu-20210916/core/crypto/aes-ccm.o
  CC [M]  /home/patrick/8821cu-20210916/core/crypto/aes-omac1.o
  CC [M]  /home/patrick/8821cu-20210916/core/crypto/ccmp.o
  CC [M]  /home/patrick/8821cu-20210916/core/crypto/gcmp.o
  CC [M]  /home/patrick/8821cu-20210916/core/crypto/aes-siv.o
  CC [M]  /home/patrick/8821cu-20210916/core/crypto/aes-ctr.o
  CC [M]  /home/patrick/8821cu-20210916/core/crypto/sha256-internal.o
  CC [M]  /home/patrick/8821cu-20210916/core/crypto/sha256.o
  CC [M]  /home/patrick/8821cu-20210916/core/crypto/sha256-prf.o
  CC [M]  /home/patrick/8821cu-20210916/core/crypto/rtw_crypto_wrap.o
  CC [M]  /home/patrick/8821cu-20210916/core/rtw_swcrypto.o
  CC [M]  /home/patrick/8821cu-20210916/os_dep/osdep_service.o
  CC [M]  /home/patrick/8821cu-20210916/os_dep/linux/os_intfs.o
  CC [M]  /home/patrick/8821cu-20210916/os_dep/linux/usb_intf.o
  CC [M]  /home/patrick/8821cu-20210916/os_dep/linux/usb_ops_linux.o
[... 96 lines cut ...]
  CC [M]  /home/patrick/8821cu-20210916/hal/phydm/halrf/halrf_debug.o
  CC [M]  /home/patrick/8821cu-20210916/hal/phydm/halrf/halphyrf_ce.o
  CC [M]  /home/patrick/8821cu-20210916/hal/phydm/halrf/halrf_powertracking_ce.o
  CC [M]  /home/patrick/8821cu-20210916/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /home/patrick/8821cu-20210916/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /home/patrick/8821cu-20210916/hal/phydm/halrf/halrf_psd.o
  CC [M]  /home/patrick/8821cu-20210916/hal/phydm/rtl8821c/halhwimg8821c_bb.o
  CC [M]  /home/patrick/8821cu-20210916/hal/phydm/rtl8821c/halhwimg8821c_mac.o
  CC [M]  /home/patrick/8821cu-20210916/hal/phydm/rtl8821c/phydm_hal_api8821c.o
  CC [M]  /home/patrick/8821cu-20210916/hal/phydm/rtl8821c/phydm_regconfig8821c.o
  CC [M]  /home/patrick/8821cu-20210916/hal/phydm/rtl8821c/phydm_rtl8821c.o
  CC [M]  /home/patrick/8821cu-20210916/hal/phydm/halrf/rtl8821c/halhwimg8821c_rf.o
  CC [M]  /home/patrick/8821cu-20210916/hal/phydm/halrf/rtl8821c/halrf_8821c.o
  CC [M]  /home/patrick/8821cu-20210916/hal/phydm/halrf/rtl8821c/halrf_iqk_8821c.o
  CC [M]  /home/patrick/8821cu-20210916/hal/btc/halbtc8821cwifionly.o
  CC [M]  /home/patrick/8821cu-20210916/hal/btc/halbtc8821c1ant.o
  CC [M]  /home/patrick/8821cu-20210916/hal/btc/halbtc8821c2ant.o
  CC [M]  /home/patrick/8821cu-20210916/platform/platform_ops.o
  LD [M]  /home/patrick/8821cu-20210916/8821cu.o
  MODPOST /home/patrick/8821cu-20210916/Module.symvers
  CC [M]  /home/patrick/8821cu-20210916/8821cu.mod.o
  LD [M]  /home/patrick/8821cu-20210916/8821cu.ko
make[1]: Leaving directory '/usr/src/linux-headers-6.1.31'
```
As for  the source the kernel was built from, its located in the builder; `sources/pi-linux-pi-6.1-sunxi`.

pyavitz, 2024-02-26:
This `6.1.31-sun50iw9` would be on official BPI images I'm sure.

jayofelony, 2024-02-27:
Hmm even with 6.1.31 it cannot find the build/ folder, I am talking about the M2 Zero this time not the M4 Zero.

pyavitz, 2024-02-27:
Is this an img you generated with the builder? All the images made using the builder come with headers installed.

jayofelony, 2024-02-27:
Yes with the builder

pyavitz, 2024-02-28:
I don't have the M2 or P2, so I honestly have no idea how well they run. But where the builder is concerned, those kernels will boot any of the SUNXI ARMV7 units that are currently listed in the builder and pulling from MAINLINE.

In short; those kernels should work across multiple units. I could tell the builder to compile for the M2 and then install the same kernel packages on the NanoPi M1.

```
FriendlyArm NanoPi M1
Ubuntu 20.04.6 LTS (6.1.77 #1 Sat Feb 10 20:36:55 EST 2024 armv7l)

ls /lib/modules/`uname -r`/build
arch    Documentation  init      kernel    Module.symvers  scripts   usr
block   drivers        io_uring  lib       net             security  virt
certs   fs             ipc       Makefile  rust            sound
crypto  include        Kconfig   mm        samples         tools
```

So when you say the headers aren't there, I don't know how that is possible. 

```
patrick@speaker:~/88x2bu-20210702$ make ARCH=arm -j4
make ARCH=arm CROSS_COMPILE= -C /lib/modules/6.1.77/build M=/home/patrick/88x2bu-20210702  modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.77'
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_cmd.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_security.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_debug.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_io.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_ioctl_query.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_ioctl_set.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_ieee80211.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_mlme.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_mlme_ext.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_mi.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_wlan_util.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_vht.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_pwrctrl.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_rf.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_chplan.o
  CC [M]  /home/patrick/88x2bu-20210702/core/monitor/rtw_radiotap.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_recv.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_sta_mgt.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_ap.o
  CC [M]  /home/patrick/88x2bu-20210702/core/wds/rtw_wds.o
  CC [M]  /home/patrick/88x2bu-20210702/core/mesh/rtw_mesh.o
  CC [M]  /home/patrick/88x2bu-20210702/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/patrick/88x2bu-20210702/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_xmit.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_p2p.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_rson.o
  CC [M]  /home/patrick/88x2bu-20210702/core/rtw_tdls.o
^Z
[1]+  Stopped                 make ARCH=arm -j4
```

pyavitz, 2024-02-28:
> Hmm even with 6.1.31 it cannot find the build/ folder, I am talking about the M2 Zero this time not the M4 Zero.

How about this. What are you trying to compile? Is it available for download so I can attempt to build it?

jayofelony, 2024-02-28:
Nexmon

pyavitz, 2024-02-28:
When you run `make config` the script checks the webs for the latest kernel rev available. Currently for 6.1.y I believe that is 6.1.79. So if you want the latest and greatest, I would build that. As for nexmon, I just compiled it on a NanoPi M1.

https://paste.debian.net/1308899/

`sudo apt install libgmp3-dev gawk qpdf bison flex make xxd autoconf texinfo`

```
source setup_env.sh
make
```

jayofelony, 2024-02-28:
I have chroot into the image I made, When I run `uname -r`  it says 6.1.31, but there is only a `/lib/modules/6.1.79/build` folder.

jayofelony, 2024-02-28:
[image]
nevermind, as usual I forgot I set this as 6.1.31

closed 2024-03-09 by pyavitz