pyavitz/debian-image-builder #102: Support for Orange Pi 5 Plus

b9AcE, 2024-03-02:
Having noted that the "Orange Pi 5"-device is already at least somewhat supported (work in progress), I wonder what would be needed to also add support for the similar "[Orange Pi 5 Plus](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-5-plus.html)"?
It has e.g. several Debian images officially from the device manufacturer (bookworm and bullseye with 6.1.x and 5.10.x kernels) as well as several other device manufacturer official and third-party Linux [distro-images](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-5-plus.html), which could aid the adding of the device to this script.
Hoping to be able to use Devuan on the device through this method.

pyavitz, 2024-03-02:
It builds. I have no clue if or how it works. https://github.com/pyavitz/debian-image-builder/pull/103

b9AcE, 2024-03-02:
I built it (on the Orange Pi 5 Plus running DietPi) with no apparent issues, then flashed it onto an SD card using balenaEtcher, then booted the Orange Pi 5 Plus from that and it all seems to work fine, including both wired and wireless networking as far as I tested it. The NVMe  on the integrated M.2 at least showed up to lsblk but I did not test that any farther.
I have not tested other devices such as Bluetooth or audio.
Thank you!

closed 2024-03-02 by b9AcE